clach04/python-tuya #3: Encryption padding issue

clach04, 2018-01-03:
Need more info., see issue #1

clach04, 2018-01-03:
@nijave

> I was also running it an issue with _pad where is tries to concatenate bytes with a string and leads to an exception

I'm definitely not seeing this with either PyCrypto nor pyaes. We need more information.

Commit dae73ad60064b1db59dab998760ce149c556787b - dumps the Python version, which may be relevant. I've seen odd byte/string behaviors with old versions of Python 3.x. Enable logging as for #1

nijave, 2018-01-04:
See example error in https://github.com/home-assistant/home-assistant/pull/11000#issuecomment-354947300

That person appears to be using python35 on Windows. I'll do some more research into this and try to figure out if it's Python version specific or based on which crypto library is used

nijave, 2018-01-04:
On my HA

```
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/tasks.py", line 182, in _step
    result = coro.throw(exc)
  File "/usr/lib/python3.6/site-packages/homeassistant/core.py", line 1031, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/usr/lib/python3.6/site-packages/homeassistant/components/switch/__init__.py", line 113, in async_handle_switch_service
    yield from switch.async_turn_on()
  File "/usr/lib/python3.6/asyncio/futures.py", line 332, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.6/asyncio/tasks.py", line 250, in _wakeup
    future.result()
  File "/usr/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/switch/tuya.py", line 83, in turn_on
    self._device.set_status(True, self._switchid)
  File "/config/deps/lib/python3.6/site-packages/pytuya/__init__.py", line 228, in set_status
    payload = self.generate_payload(command, dps_id=switch)
  File "/config/deps/lib/python3.6/site-packages/pytuya/__init__.py", line 163, in generate_payload
    json_payload = self.cipher.encrypt(json_payload)
  File "/config/deps/lib/python3.6/site-packages/pytuya/__init__.py", line 41, in encrypt
    raw = self._pad(raw)
  File "/config/deps/lib/python3.6/site-packages/pytuya/__init__.py", line 71, in _pad
    return s + (self.bs - len(s) % self.bs) * chr(self.bs - len(s) % self.bs)
TypeError: can't concat str to bytes
```

clach04, 2018-01-04:
@nijave hopefully resolved in commit fb1c28993d10a85ec2220a24a51323ec4feacccd

clach04, 2018-01-04:
Turns out the reason I didn't see this was I had not ran the Python 3 and pycrypto/pycryptodome combination (only py3 and pyaes).

nijave, 2018-01-04:
Just curious, any reason you're using your own padding methods instead of the ones packaged with pycrypto? See https://github.com/nijave/python-tuya/commit/3c94c9dab141c746a6e36ed80a19d8ea99232a20

clach04, 2018-01-04:
I don't have that with the version of PyCrypto I have installed.

    Python 2.7.13 |Anaconda 4.3.1 (32-bit)| (default, Dec 19 2016, 13:36:02) [MSC v.1500 32 bit (Intel)] on win32
    Type "help", "copyright", "credits" or "license" for more information.
    Anaconda is brought to you by Continuum Analytics.
    Please check out: http://continuum.io/thanks and https://anaconda.org
    >>> import Crypto
    >>> Crypto.__file__
    'C:\\ProgramData\\Anaconda2\\lib\\site-packages\\Crypto\\__init__.pyc'
    >>> Crypto.version_info
    (2, 6, 1, 'final', 0)
    >>> from Crypto.Util import Padding
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    ImportError: cannot import name Padding

Otherwise this would be much more elegant :-(

I took your branch and then merged with master into https://github.com/clach04/python-tuya/tree/3-encryption-padding - but it doesn't work for me.

Can you confirm the master branch now works for you? Then we can close this one down.

nijave, 2018-01-04:
I'll go ahead and do some testing with pycrypto (the older one I think 2.6.1), pycryptodome (3.7.1 I think is what's on HA), and pyaes on both Python2 & 3. If I have time I'll do Linux, too, although I wouldn't imagine there are any issues there based on the changes. That should give us some better coverage

clach04, 2018-01-12:
Resolved by pull request #9

closed 2018-01-12 by clach04