clach04/python-tuya #11: Hang when TCP connection is interrupted

nijave, 2018-01-18:
The module gets stuck when the TCP connection is interrupted (it blocks waiting for response). Not sure what causes the TCP interruptions--I suspect network issues due to the high level of interference and outlets on 2.4Ghz.

Possibly add a send/receive timeout?

clach04, 2018-01-18:
@nijave have you tried tried `socket.setdefaulttimeout()`? Its not something I've seen with my hardware but I can see how this could happen.

nijave, 2018-01-18:
Yeah, that's what I'm using right now

clach04, 2018-01-19:
Cool, if you can make a PR for this that would be awesome. In terms of reporting back to the client, I'm thinking we have something like:

    class BaseException(Exception):
        '''Base python-tuya exception'''
        def __repr__(self):
            return repr(self.__dict__)

        def __str__(self):
            return repr(self.__dict__)

    class CommsError(BaseException):
        """Communication Error"""
        pass

    raise CommsError('timed out on send %r' % realexception.info....)

That way clients can check one thing, the probably don't care about the details (other than for potential debug logging). If that assumption is incorrect we can expand the exceptions. We need to normalize across operating systems (and Python versions), hence suggesting the wrapper. Be interested to get your take on this,

closed 2018-02-12 by clach04