clach04/python-tuya #13: Older Python 3 json problems -  TypeError: the JSON object must be str, not 'bytes'

clach04, 2018-01-25:
Under Python 3.5.4 (and maybe 3.4.3, see issue #10), get:

    ERROR: test_set_status (__main__.TestXenonDevice)
    ----------------------------------------------------------------------
    Traceback (most recent call last):
      File "tests.py", line 116, in test_set_status
        result = json.loads(result)
      File "/opt/python/3.5.4/lib/python3.5/json/__init__.py", line 312, in loads
        s.__class__.__name__))
    TypeError: the JSON object must be str, not 'bytes'
    ======================================================================
    ERROR: test_set_timer (__main__.TestXenonDevice)
    ----------------------------------------------------------------------
    Traceback (most recent call last):
      File "tests.py", line 106, in test_set_timer
        result = json.loads(result)
      File "/opt/python/3.5.4/lib/python3.5/json/__init__.py", line 312, in loads
        s.__class__.__name__))
    TypeError: the JSON object must be str, not 'bytes'

Looking at this, it looks like a json library bug, json.loads() should accept bytes.

clach04, 2018-01-25:
@Exilit  can you take a look at this, is this what you are seeing? What happens if you open a Python3 prompt and enter:

    import json
    json.loads(b'{}')
    json.loads('{}')

@mike-gracia can you confirm what you are seeing on your raspbian install?

Thanks!

Exilit, 2018-01-25:
Here is what I get with my portable Python 3.4.3 installation:
```
>>> import json
>>> json.loads(b'{}')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<PATH_TO_PATHON>\lib\json\__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'
>>> json.loads('{}')
{}
```

Seems really to be a bug in the json library. Python 2.6.4, Python 2.7.14 and Python 3.6.4 all accept both bytes and str.
I hope it is not related to be a portable installation. Not sure, where it is from.

mike-gracia, 2018-01-25:
3.6.4 works and 3.5.3 fails. See below
`
admin@DESKTOP-GGK2PSF MINGW64 ~/projects/tuya/python-tuya (master)
$ python --version
Python 3.6.4

admin@DESKTOP-GGK2PSF MINGW64 ~/projects/tuya/python-tuya (master)
$ python tests.py
...
----------------------------------------------------------------------
Ran 3 tests in 0.004s

OK
`

The output on the raspberry pi running 3.5.3 fails below.
`
pi@pi:~/projects/python-tuya $ python3 --version
Python 3.5.3
pi@pi:~/projects/python-tuya $ python3 tests.py
EE.
======================================================================
ERROR: test_set_status (__main__.TestXenonDevice)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "tests.py", line 116, in test_set_status
    result = json.loads(result)
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'

======================================================================
ERROR: test_set_timer (__main__.TestXenonDevice)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "tests.py", line 106, in test_set_timer
    result = json.loads(result)
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'

----------------------------------------------------------------------
Ran 3 tests in 0.012s

FAILED (errors=2)
`

Exilit, 2018-01-25:
@mike-gracia @clach04 I think the main question is, whether we want to work-around that or just restrict the python versions. I think targeting the raspberry pi is important, so maybe we should add something like the following:

`result = json.loads(str(data, "utf-8"))`

Exilit, 2018-01-25:
@mike-gracia Just saw your commit. Looks also good.

clach04, 2018-01-27:
Resolved by #14

closed 2018-01-27 by clach04