clach04/python-tuya #19: Key needed?

Apfelcheck, 2018-02-25:
Hi,

do I need a key if I want to develop with your API? 

Regards

Waldemar

Exilit, 2018-02-25:
What kind of key do you mean? There isn't something like an api key needed for this library. But in order to be able to use it you need the 'local key' of your device. There are a few ways to retrieve it. For more information see https://github.com/clach04/python-tuya/wiki

Apfelcheck, 2018-03-01:
Thanks, because I have read it something fast and I found the word "key" so
I thought - without this you couldn't really start. I need the
current/power data from my switch. What is the best way to get this data?

Regards

Waldemar

clach04, 2018-04-21:
Exilit's comment points to the place to get this information. Closing.

closed 2018-04-21 by clach04