clach04/python-tuya #20: Key does not get recognized/throws error

Darkyputz, 2018-02-26:
Hello...

Since i dont get any node or c# working on my device i want to run the tuya controll logic on it want to try the python now...
But the code provided here is not totally clear to me. (python noob)
In the code it states:
d = pytuya.OutletDevice('DEVICE_ID_HERE', 'IP_ADDRESS_HERE', LOCAL_KEY)
and when i fill that with my values all over the code like 
d = pytuya.OutletDevice('02200506600104944eb9', '192.168.178.65', 78382189fd5087ba)
it throws a fit at the ba at the end of the local key which i know is correct
And as a quick question on the side...do i have to call this code with paramteters to get it do something or is a dumb code just for showing general functionality?
Casue is would not be able to alter it to get on/off/toggle into it

Thx for any thoughts

kristjaneerik, 2018-02-28:
The issue might be that you didn't put `78382189fd5087ba` in quotes, e.g. `'78382189fd5087ba'`.

It would be helpful to see the actual error message that you get.

Exilit, 2018-02-28:
@kristjaneerik is right, the key needs also to be passed as str.

The other question I don't quite understand. Which code are you talking about? 
The code in the README.md is example code that should work as is.

I started working on a command line tool called tuyactl using this library, but could not finish it so far. That would allow you to control a tuya device via command line.

Darkyputz, 2018-02-28:
Hello and good morning...
My error without the quotes looks like this:
`C:\Users\Darkyputz>python f:\TuyaPlug\python-tuya\tests.py
  File "f:\TuyaPlug\python-tuya\tests.py", line 99
    d = pytuya.OutletDevice('02200506600xxx944eb9', '192.168.178.65', 78xxx189fd
b087ba)

     ^
SyntaxError: invalid syntax
`
and with the quotes like this:

> Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Alle Rechte vorbehalten.

C:\Users\Darkyputz>python f:\TuyaPlug\python-tuya\tests.py
  File "f:\TuyaPlug\python-tuya\tests.py", line 99
    d = pytuya.OutletDevice('02200506600194944eb9', '192.168.178.65', 78c82189fd
b087ba)

     ^
SyntaxError: invalid syntax

C:\Users\Darkyputz>python f:\TuyaPlug\python-tuya\tests.py
ERROR:root:json data not the same: {"devId":"02200506600194944eb9","uid":"022005
06600194944eb9","t":"1519828256","dps":{"1":true}} != {"dps":{"1":true},"uid":"D
EVICE_ID_HERE","t":"1516117564","devId":"DEVICE_ID_HERE"}
FFERROR:root:json data not the same: {"gwId":"02200506600194944eb9","devId":"022
00506600194944eb9"} != {"devId":"DEVICE_ID_HERE","gwId":"DEVICE_ID_HERE"}
F
======================================================================
FAIL: test_set_status (__main__.TestXenonDevice)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "f:\TuyaPlug\python-tuya\tests.py", line 121, in test_set_status
    self.assertEqual(result['test_result'], "SUCCESS")
AssertionError: 'FAIL' != 'SUCCESS'
- FAIL
+ SUCCESS
======================================================================
FAIL: test_set_timer (__main__.TestXenonDevice)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "f:\TuyaPlug\python-tuya\tests.py", line 110, in test_set_timer
    self.assertEqual(result['test_result'], "SUCCESS")
AssertionError: 'FAIL' != 'SUCCESS'
- FAIL
+ SUCCESS
======================================================================
FAIL: test_status (__main__.TestXenonDevice)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "f:\TuyaPlug\python-tuya\tests.py", line 130, in test_status
    self.assertEqual(result['test_result'], "SUCCESS")
AssertionError: 'FAIL' != 'SUCCESS'
- FAIL
+ SUCCESS
----------------------------------------------------------------------
Ran 3 tests in 0.007s

FAILED (failures=3)

>

Darkyputz, 2018-02-28:
Sorry for the strange look of the answer, but i dont really know why it does it in bold and that big...

Darkyputz, 2018-02-28:
And to answer your question...
The sample code in the readme...
I have no idea how to use that...
Simply creating a file with .py as extension with that lines in it does nothing...just as an example how python nooby i am...

Darkyputz, 2018-02-28:
Hello...
Thx for the replies...
I think i found my culprit after reading your comments...
The demo now seems to toggle the switch now...but i had to comment out the part below the toggle since mine is just a one outlet only...
Now a question again...
It says import pytuya on top...
This is right now all on my windows machien but the final plan is to run that all semi automatik on my mips router...so what/where is this pytuya coming from that is called?

Thx Jan

Darkyputz, 2018-02-28:
Ok...after playing around, i got it working on my mips based router with fresh compiled python 2.7
JUST AWESOME!!!

closed 2018-02-28 by Darkyputz