clach04/python-tuya #21: AttributeError: 'str' object has no attribute 'decode'

theit8514, 2018-03-14:
Using pytuya with tuya-homeassistant (python 3.6.4) and I get this error when toggling on/off the switch:
```
  File "/config/deps/lib/python3.6/site-packages/pytuya/__init__.py", line 275, in status
    result = json.loads(result.decode())
AttributeError: 'str' object has no attribute 'decode'
```

mike-gracia, 2018-03-22:
Have you tried removing the .decode() method from that line? Not totally sure how that code path works with the encryption but if result is a string already you do not need the decode method.

theit8514, 2018-03-22:
This has been working for a while now with no issues:
```
            log.debug('decrypted result=%r', result)
            if not isinstance(result, str):
                result = result.decode()
            result = json.loads(result)
```

clach04, 2018-04-21:
@theit8514 what are you using for crypto? I'm trying to work out which code path you are dropping through. I'm going to guess PyCrypto (or PyCryptodome) based on a quick code inspection (AESCipher wrapper class will return a string).

theit8514, 2018-04-22:
These are the crypto modules installed (from the `homeassistant/home-assistant` docker container):

$ pip list | grep crypto -i
asn1crypto (0.24.0)
cryptography (2.1.1)
pycryptodome (3.4.11)
pycryptodomex (3.5.1)
python-jose-cryptodome (1.3.2)

clach04, 2018-04-22:
Thanks for confirming. So this is one of those cases where we are missing a test for this code path. It would be much better if we have the correct code than a protective check/cast. If anyone has time to work on a unittest that would be awesome.

left4taco, 2018-10-29:
I guess you (or the author of the home-assistant plugin) fixed the `pytuya` version to 6.0.0 in the script. Change it to 7.0.2 would solve this problem.