clach04/python-tuya #25: Wiki usb power strip

BillSobel, 2018-04-25:
A comment on the wiki notes: NOTE dps.6 looks like a timer 

Actually this is defined as switch_all in the schema and should do an all on or all off if a 1 or 0 is set. I ordered one to verify. Will update when it arrives.

clach04, 2018-04-25:
Thanks @BillSobel for taking the time to check this out and comment. I'm guessing you mean the end of https://github.com/clach04/python-tuya/wiki#sm-s0301-us versus the contents of the schema https://github.com/clach04/python-tuya/wiki/DeviceSchemas#wuudi-sm-s0301-us - I think you are correct. I don't have access to my device to test at the moment though :(

The wiki should be editable if you find out before me.

BillSobel, 2018-04-25:
@clach04  Excuse a (possibly) very stupid question.  How do I edit the wiki.  I didn't see any options (do I need to sync the file local, edit, and sync?)  I traveling but back home on Friday, there should be a pile of devices Amazon has delivered while I was gone :)

clach04, 2018-04-26:
@BillSobel not a dumb question, I made a dumb comment! I had the wiki locked to collaborators only, so I was wrong (in those situations one can clone the repo (underneath the pages link on the right) but its a bit of a pain!

Unlocked wiki 🤞 

Have fun with the new toys over the weekend :-)