clach04/python-tuya #26: AttributeError when decoding

tadly, 2018-05-12:
I've been playing with this and (at least with python 3) the following line causes an `AttributeError: 'str' object has no attribute 'decode'`

https://github.com/clach04/python-tuya/blob/01c123f899e0ded10554fb0a6fc5e64d20176d30/pytuya/__init__.py#L274

As I'm not familiar with the code yet I quickly fixed it locally with:
```
if not isinstance(result, str):                                    
    result = result.decode()                                       
result = json.loads(result)
```

Not sure how python2 behaves with all that.

I'm rather busy lately so I don't have the time to investigate further and come up with a proper fix.

clach04, 2018-05-20:
@tadly when are you seeing this? When running the testsuite? What (sub-)version of Python?

tadly, 2018-05-20:
@clach04 nope, not testsuite. Was a live test.

As the power strip wasn't working stable enough when disabling internet access on the router, I sent it back already. So I can't do more testing.

IIRC it was with both, python 3.5.3 and 3.6.5

PR #28 (2018-06-23): Add decode check for Python 3
Closes #26
commit e407bd9b (2018-06-23): Add decode check for Python 3

Closes #26
closed 2018-06-23 by clach04