clach04/python-tuya #27: Color temperature is broken

DeltaWhy, 2018-05-25:
Looks like this line is the problem - it's sending the string `'white'` instead of the variable `white`:
https://github.com/clach04/python-tuya/blob/fe35ba3a5dfd298d25b4f62f89064f17d91f120a/pytuya/__init__.py#L386

clach04, 2018-05-25:
@DeltaWhy interesting find, annoying if the keys are different :(

Can you dump the schema for the bulb? See https://github.com/clach04/python-tuya/wiki/DeviceSchemas 

I'm going to guess we need another model/class in the module to handle this case. I don't have any bulbs myself (only switches) so we're relying on a collaborative effort on this one.

CC @samuscherer on this as they may be interested.

samuscherer, 2018-05-25:
Very interesting, but it shouldn't be a problem at all to implement another class/model for your kind of bulb. You can assign this issue to me @clach04 and I'll take a look at it asap.

DeltaWhy, 2018-05-25:
Here's the schema (this was under `.productBeen[0].schemaInfo.schema` as an escaped string, I did my best to un-mangle it):
```
[
  {
    "mode": "rw",
    "code": "led_switch",
    "name": "\u00e5\u00bc\u20ac\u00e5\u2026\u00b3",
    "property": {
      "type": "bool"
    },
    "id": 1,
    "type": "obj",
    "desc": ""
  },
  {
    "mode": "rw",
    "code": "bright_value",
    "name": "\u00e5\u00bc\u20ac\u00e5\u2026\u00b3",
    "property": {
      "unit": "",
      "min": 25,
      "max": 255,
      "scale": 0,
      "step": 1,
      "type": "value"
    },
    "iconname": "icon-dp_sun",
    "id": 2,
    "type": "obj",
    "desc": ""
  },
  {
    "mode": "rw",
    "code": "temp_value",
    "name": "\u00e5\u00bc\u20ac\u00e5\u2026\u00b3",
    "property": {
      "unit": "",
      "min": 0,
      "max": 255,
      "scale": 0,
      "step": 1,
      "type": "value"
    },
    "iconname": "icon-dp_light",
    "id": 3,
    "type": "obj",
    "desc": ""
  }
]
```
And here's the Amazon item: https://www.amazon.com/gp/product/B075GGXNZ1/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

I dumped the schema by registering the bulb with the 'Smart Life' app on a rooted phone, but strangely enough the API doesn't seem to work with that app (I did update the localKey). I re-registered through the 'TuyaSmart' app and now it works again. Guessing there's different servers for the different branded apps?

clach04, 2018-05-26:
@samuscherer for some reason I can only assign @DeltaWhy or myself, not sure why.

@DeltaWhy thanks for the information! I can confirm from discussions on the javascript project that kick started this python project that the backend servers are completely separate (it also matches my experiments too), I tend to only use the smart life app to dump the schema.

BillSobel, 2018-05-26:
The back end servers are the same.  The only difference is the keys used to make the requests. Each key set is a unique silo on the back end, but running in same AWS servers.

BillSobel, 2018-05-26:
Can I sugggest that specific hardware should not be modeled but rather that the different data point types should be modeled.  There is no reason one could not ship a powerstup with 5 outlets and an rgb indicator. Modeling DPs supports any device generically.

samuscherer, 2018-06-05:
Sorry that you guys haven't heard anything from me - I'm currently pretty busy and will be for a few more days but I haven't forgotten this issue and will definitely be looking into it.

ch491, 2018-06-20:
I am a NOOB to Nodeservers and Polyglot. @DeltaWhy What node server are you using to control the Lumiman bulbs? I have one to test before I purchase many more. I am using ISY994IZW and want to control either the Lumiman or TP-Link Tunable White LED bulbs.  Any pointers for a Noob would be welcome. TY in advance.

clach04, 2018-06-21:
@ch491 this is a Python based library, not Node. From what I can gather the ISY994IZW is a Z-Wave based device, this library is only for WiFi based devices.

BillSobel, 2018-06-21:
The ISY supports zwave but also other devices via a plugin architecture
based on node.  A shim between that and the library would be needed, but
doable for what the other poster was looking for.

ch491, 2018-06-21:
Thank you both, I was going off of @DeltaWhy 's post as it sounded like he was using the python GitHub project Ployglot (a nodeserver) that is the "shim" between the ISY home automation controller (zwave,Inston,and some IP control) to control Wifi Bulbs like LiFX etc using the MQTT protocol. Looking to see is a shim (python) has been created for the Lumiman (Wifi/MQTT) LED bulbs.