clach04/python-tuya #30: RecTec Pellet Smoker Controller Support

SDNick484, 2018-07-17:
Clach04, first let me say thank you for all the work you've put into python-tuya; it's already been incredibly helpful for me.  I recently acquired the new WiFi controller for my RecTec pellet smoker and after some research I found it leverages the Tuya platform which led me to both your project and Tuyapi.  So far I've been able to leverage your status function to get some good info from the smoker (state, temperatures, etc.) however I haven't been able to change its state (i.e. turn it on or off) with the set function (which I realize is designed for other devices - I was just hoping I might get lucky).  I was hoping you might have some insight into how I might be able to do so (or if I'll need to figure out how to leverage the MQTT interface instead).  The scripts I've been working on so far are available [here](https://github.com/SDNick484/rectec_status/).  I hope you don't mind, but I've added your [discover devices script](https://github.com/codetheweb/tuyapi/issues/5#issuecomment-348799963) as it greatly simplifies finding the device ID.

BillSobel, 2018-07-17:
Can you post the schema for the device?

clach04, 2018-07-17:
@SDNick484 - that is awesome! It fills me with so much joy to know that someone is using something I wrote to  :fire: burn  :fire: things! :grinning: The Alexa skill is a really nice addition.

reverse engineered some information already and posted it in your readme https://github.com/SDNick484/rectec_status/ - As @BillSobel posted, getting the device schema will help grealy here. This would mean taking a different app to register the grill (temporarily) to dump the schema. See https://github.com/clach04/python-tuya/wiki/DeviceSchemas

What have you tried already for turning on/off?

The other avenue to explore would be to trace the official app using tcpdump and then decrypt the packets (you may already have seen the code I posted code for that). I recommend holding off on this but I'm not sure when I'll get chance to respond so dumping what I know during my lunch break, the following should work:

http://www.tcpdump.org/tcpdump_man.html (you may need root, been a while since I did this):

     exec tcpdump -i any -w /sdcard/tcpdump.out  host  grill.ip.address.here

RE MQTT - I do NOT think that is an option. I don't think any of my devices support local MQTT. From what I can gather that is done via the cloud service (which this project does not have support for).

No problem bundling the discover code. At some point it would be nice to get that integrate into the library (PR's welcome :wink:).

SDNick484, 2018-07-18:
Thank you both for the quick responses; I will work on getting the schema tonight.  I've only had the new controller for a week so I haven't had a chance to do much exploration (yet!) however I have already taken a couple tcpdumps during a couple cookouts.  I'll get those posted too.  Haven't tried registering the device with other apps, but that was on my to-do list when I get some time.

BillSobel, 2018-07-18:
Googled this real quick and found the below project, they realized its
yours, lol.  Looks like DP1 is the on/off control...  If you can give me
the product key you mention being broadcast, I can look up the full schema
(the below is a partial schema, not the full tuna one which are more
descriptive).  I have Tuya developer keys and can call the cloud API, the
productKey is a shared key for all devices of the same type.

Bill

SDNick484, 2018-07-18:
Hi Bill, the productKey is "1CwNbNLE8r70WR7t".  I was able to get the schema and will post it on the product page shortly.  I am also working on getting it formatted correctly but am having a little trouble.  I used the Packet Capture + TuyaSmart app method.

impala454, 2020-01-20:
I know this is an ancient ticket but I bought a RecTec Stampede back in November and am trying the darndest to get the data from it and into my home automation system.  I honestly don't necessarily even care to have control so much as get the data from the grill like current temperatures and such.  I have looked at @SDNick484 's stuff but can't get any of it to work.  The broadcast packet I see is on port 6667 and is some kind of binary encoded data and not readable json.  When I wireshark the grill, the only packets I see are that broadcast and a TLS encrypted TCP conversation whenever I use the app to control the grill.  I was able to peruse around the app and find a `TUYA_SMART_APPKEY` and `TUYA_SMART_SECRET` but not sure how to use those.

Any help would be greatly appreciated.  I love the grill but am annoyed that their app's temperature alarms don't work and would like to use my home built NodeRED/Grafana setup to monitor.

clach04, 2020-01-20:
@impala454 I'm out of date with current protocol - I believe there maybe a new one (I sometimes see emails on the https://github.com/codetheweb/tuyapi ticking system. If the original protocol is in use then the port doesn't look correct - it should be possbly to connect to 6668 and active poll it.

Check out the node implementation. There were some notes in https://github.com/clach04/python-tuya/wiki/DeviceSchemas too - again, I'm not clear on current status.

If you have the key then you should in theory have what is needed to decrypt the payload - at least for when polling. I've no idea about when these devices broadcast.

impala454, 2020-01-20:
@clach04 much appreciated.  I've checked a lot more and looked at the tuyapi but seems the capturing the device id doesn't work like it used to and even if you succeed you can then only control it from the new device.  I sniffed the broadcast packets again and noticed there's no change in the payload when turning the grill on and seeing temperature changes.  Looks like what I want to do (just grab the temperature data) isn't possible now.  From all the sniffing I've done, it appears the broadcast is purely for discovery (as you previously discovered), then the TLS handshake is between the phone app and grill directly over the LAN, then it seems the temperature data updates to the app come straight to the app from the cloud.

SDNick484, 2020-02-14:
@impala454 I am getting back into my project and will try to take a look soon to see what's cha nged. You could definitely get temp data previously over local TCP previously (although you are correct that the app uses MQTT data from the cloud for that). I am hoping to get some Home Assistant integration in place but am pressed for time. Will hopefully have a chance to make some progress in the next few weeks.

SDNick484, 2020-02-14:
@impala454 I just checked my code and it's still working for me on my RT-680 WiFi controller.  I'm wondering if the Stampede is using a newer chip/protocol. 

Does this script work for you: https://github.com/codetheweb/tuyapi/issues/5#issuecomment-348799963

impala454, 2020-02-15:
@SDNick484 am unable to test at the moment but will try soon.  I'm very interested in running this to ground, as RecTec support has not been forthcoming in trying to get me any data.

impala454, 2020-02-15:
@SDNick484 the script you pasted is nearly identical to what I've been trying, and no dice.  I don't see anything on port 6666.  When I wireshark the grill I still just see the packets on 6667 I mentioned above, which appear to be encrypted.  Here's a sample (note that the contents do not change, even with the grill on):
```000055aa0000000000000013000000ac00000000aa613a3bdc1e4a5a5c8504d88beb4c44b7fcece0445a1705da4358c12c063fc90fce36c6c7e17c13b3f3db89d740c01ec77675ec78d575e4d20f57eda05ae5bfcb5924bb8792fe5eae7c9040e5921a98017f2170860421e6fc66f0a1ade138915fde850cd157270c5f3d1ab5ae90b300974ebcfdbac62e220d35be537c43de671c8343bc25f3403c0c4c3db59350d34ae0bd1f2cbad705886d4ed8eed5ab9eb9475641140000aa55```

impala454, 2020-02-15:
Another note, the TCP traffic is definitely TLS encrypted, and going to AWS at regular intervals, or quick bursts while I'm using the app.

SDNick484, 2020-02-15:
Interesting, yeah, they must have made changes as that's definitely not what I'm seeing.   Would be interesting to know which models/controllers that affects.  I'll see if any of my other devices are on the newer firmware and will try to do some digging.  If you haven't already, I'd recommend checking out this thread: https://github.com/codetheweb/tuyapi/issues/175

impala454, 2020-02-15:
Minor updates:  I tried the following and it seems I get nothing back until I change `version` to `3.3`.  So I'm guessing that's the version my grill has.  Note the return is encrypted or at least serialized in some way and unreadable.

```javascript
const TuyAPI = require('tuyapi');

const device = new TuyAPI({
  key: 'xxxxxxxxxxxxxxxx',
  ip: '10.20.30.220',
  version: 3.3});

(async () => {
  await device.find();

  await device.connect();

  let status = await device.get();

  console.log(`Current status: ${status}.`);

  device.disconnect();
})();
```

Another update is I was able to de-compile the RecTec app and retrieve two interesting values:  `TUYA_SMART_APPKEY` and `TUYA_SMART_SECRET` from the `AndroidManifest.xml` file. I've attempted to use the latest tuya-cli link tool to connect to the grill with no success.