clach04/python-tuya #31: Sometimes "Connection Refused" occurs

LucidSkyWalker, 2018-07-20:
Hi together,

I am using python-tuya in my Snips Home Automation.
It works fine, but sometimes I get a "Connection Refused".
I know that it is blocked when the App is open (you think there is a possiblity work around?),
but also if the Mobile App is closed I sometimes get a "Connection Refused".

When I reopen the App, turn off and on the light switches and then close the app
it is working again...

Do you have any idea or solution for this?

Thank you for your python library!!! Most of the time it works really nice :))

BillSobel, 2018-07-21:
The devices only accept one TCP connection, if the connection is used then
another connection can not be made.  You would need custom firmware to fix
this.

One option for the most compatibility would be to firewall your mobile
device so it can *not* talk to the devices directly, it will automatically
fall back to MQTT.  This way your HA system can always rely on TCP being
available.

In my home seer plugin I support MQTT as well, for exactly this reason.  I
hope at some point this project supports it (the work to do TCP covers MUCH
of the MQTT requirements, including encryption and what not).

Cheers,
Bill

left4taco, 2018-10-29:
@BillSobel 
Hi Bill, 
I also have this connection refused problem with my plugs. So you mean, the mobile application Tuya has two methods to communicate with the devices:
1. via TCP if Tuya app and plug are in the same LAN. 
2. via a MQTT broker provided by Tuya. App publishes a message to the broker topic which is listened by the plug. 

Is that right? If so, is there any possible way to configure the MQTT broker address inside the plug? I assume it's fixed in the firmware.

BillSobel, 2018-10-29:
Correct but also correct it’s hard coded in the firmware. 
Bill