clach04/python-tuya #32: Hola, I wrote an implementation of codetheweb's stuff too

Supermortal, 2018-08-09:
Hey, I just wanted to post this to you guys, and see what you thought. I re-wrote the tuyapi stuff as a project to help me learn Python, and I figured I'd post it to you guys. I don't want to step on toes or anything. https://github.com/Supermortal/python-tuya-oittm

clach04, 2018-08-09:
@Supermortal neat! I have to confess I've not had/made time to work on this project recently and you've done a few things I was hoping to get into this implementation (e.g. discovery is something I wrote and got working but didn't get into this library).

I only had a few mins to look at your implementation. So these are random comments.

1. I couldn't get this running under Python 2.7.13 - due to the F-String (PEP-498) usage. I noticed six was a dependency so I suspect you are aiming for support of both.
2. There are a lot of dependencies, but I did get a quick demo running with Python 3.6 and discovery. Nice job!
3. I like the use if async, great idea!
4. Nice to see you have the start of tests for most aspects
5. Use of a dict for device support is neat. Be interesting to see if you can expand that to other devices
6. The OITTM_REVERSE_HUMIDIFIER_MAP in device_maps is hard coded, recommend creating that dynamically based on OITTM_HUMIDIFIER_MAP to avoid maintenance overhead (and potential for bugs)

You're welcome to take anything useful from this project (thanks for the acknowledgment BTW, its appreciated), thanks for sharing. Very cool. It would be nice if we could merge the projects some how but I don't have a lot of time these days to work on this so its not high on my todo list.

Supermortal, 2018-08-09:
Thanks for commenting. Yeah, I'd be down to merge, and I can look at the stuff you mentioned for sure.

Supermortal, 2018-08-11:
Would you mind if I pushed this up to PyPi, so I can start development on my Home Assistant component? If/when we do decide to merge code bases, I'd be happy to deprecate the PyPi package, I'd just like to get started on that soon. @clach04

clach04, 2018-08-12:
@Supermortal go for it! Thanks for taking the time to work on it.

cmille34, 2018-08-22:
This is exciting news!  I still get random timeouts with the tuya.py in Home Assistant. Its great to see someone continuing to develop this for that platform.  Thanks for your efforts!