clach04/python-tuya #33: socket timeout error

Minichota, 2018-08-13:
When trying to run the example, I get this traceback:
Traceback (most recent call last):
  File "C:/Users/Minichota/PycharmProjects/Global_Automation/lights.py", line 4, in <module>
    data = d.status()  # NOTE this does NOT require a valid key
  File "C:\Python\lib\site-packages\pytuya\__init__.py", line 255, in status
    data = self._send_receive(payload)
  File "C:\Python\lib\site-packages\pytuya\__init__.py", line 165, in _send_receive
    s.connect((self.address, self.port))
socket.timeout: timed out

I hope you can help me or guide me through to the solution

clach04, 2018-08-14:
@Minichota  is this tuyapi issue 57 - https://github.com/codetheweb/tuyapi/issues/57 ? If so I'm going to hazard a guess that something has that device open already (https://github.com/codetheweb/tuyapi/issues/57#issuecomment-412610094 indicates that maybe the phone app has it open).

This is one of those "common knowledge" things where we know the devices can only handle one active connection at a time but to know this you either need to have been burnt by this or read alot of the history comments on issues in this and the https://github.com/codetheweb/tuyapi/issues/ project (https://github.com/codetheweb/tuyapi/issues/5 has a ton of information). This has sparked me to create a FAQ page in the wiki. I just need to get this into an entry there https://github.com/clach04/python-tuya/wiki/FAQ :-)

Minichota, 2018-08-14:
well, I'm quite sure that the device isn't connected to by multiple things as when the indicator light is blinking it is disconnected.

Minichota, 2018-08-14:
I've also unplugged the device and replugged it.

clach04, 2018-08-14:
@Minichota if the device light is blinking, then the device is not on the network. This project expect the device to already be provisioned and on the network. I'm guessing this isn't working with https://github.com/codetheweb/tuyapi/ either?

Minichota, 2018-08-14:
So are you saying that I should run it through the setup.md example you gave again? @clach04

Minichota, 2018-08-14:
I also haven't tried to use the other tuya api as I don't have a compiler for that language.

clach04, 2018-08-14:
@Minichota you mean "Key extraction", yes. You'll need to provision on your network and extract the key (using your preferred method). Unfortunately it looks like the guide I linked to was updated 6 days ago to do provisioning with the CLI tool so you either need to install Node to follow the current guide or read the older guide https://github.com/codetheweb/tuyapi/blob/9c835da1139a3436f43b359b86abeb3529ae90a1/docs/SETUP.md

Minichota, 2018-08-14:
@clach04 Do you mean to do this if I wasn't able to extract the information?

clach04, 2018-08-14:
@Minichota  I don't understand the question. If the light is blinking that means its not connected to the network and its entered provisioning mode. I've updated https://github.com/clach04/python-tuya/wiki and copied the old readme.

From your description it sounds like you need to provision the device. This will change any key(s) you may have already extracted.

Minichota, 2018-08-14:
I have connected to it using the 'old' method you posted and I have fixed the blinking light issue and its simply when I run the example you provide in the README I get a socket.timeout error as I pasted above. I have access to the all the data from the key extraction and it should technically be connected so I'm not sure.

Minichota, 2018-08-14:
@clach04 If I up the connection timeout to 50 seconds, I get this error: `TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond`

clach04, 2018-08-14:
@Minichota so definitely a network problem. I'll close this down. If its any consolation I had a similar problem once. I was super tired 😴  and my laptop connected to an open WiFi network and I couldn't work out why I couldn't access any devices on my LAN!

closed 2018-08-14 by clach04

codetheweb, 2018-08-14:
@clach04 by the way, the old setup instructions can be found [here](https://github.com/codetheweb/tuyapi/blob/master/docs/SETUP_DEPRECATED.md).