clach04/python-tuya #34: PyPi package

clach04, 2018-08-31:
Related to Issue #24.

If any one is interested and has not worked on PyPi support before, I have a sample project at https://github.com/clach04/x10_any (sadly I don't have "how to publish" notes there, from memory I used twine but I did bookmark https://nosarthur.github.io/coding/2018/07/01/pypi.html which has a note on how to use a Markdown formatted readme that I mean to try).

Random thoughts:

   * obviously need a setup.py :) (see x10_any)
  * Need a version string to be added (see x10_any)
  * pycrypto dependency may be a problem for Home Assistant (see pycryptodome)

clach04, 2018-09-01:
https://pypi.org/project/pytuya/ posted by https://github.com/sean6541 - current version 7.0, so to avoid issues initial version needs to be (at least) 7.0.1 to ensure an upgrade will work.

Exilit, 2018-09-01:
My biggest problem is currently the package name :-) as pytuya is already used by @sean6541.

We could definetly ask sean6541 to hand over the maintenance to us. But I think there might already be software out there which uses the package and I don't want to continue the version numbering (it's already 7.0).

What do you think, @clach04 ? Do you have any preferences?

clach04, 2018-09-28:
@Exilit no strong preferences on my part. What are your thoughts on the versioning, what would you like to see? I was about to add a version to the module then saw your comment, so I held back :)

clach04, 2018-10-09:
@Exilit new version posted to pypi, under pytuya name with current code.

Outstanding things to resolve:

 * readme in markdown format doesn't look correct in PyPi - this may be due to the old version of distutils I used in packaging; `Python36-32\lib\distutils\dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'`
 * dependency information

Currently the setup.py file has a dependency on pyaes. This works things that are not perfect about this setup:

1. it ignores PyCrypto/PyCryptodome - that dependency will be present for some installations (e.g. Home Assistant #24 requires it). However installing pyaes and then finding PyCrypto first works as expected, viz. PyCrypto is used in preference
2. the pyaes version on PyPi is old and has partial Python 2.x support, there is an outstanding PR (disclosure its my PR) for this https://github.com/ricmoo/pyaes/pull/13 amongst other open PRs

Leaving open for discussion for a while.