clach04/python-tuya #37: .status() occasionally restarts device

glentakahashi, 2018-09-22:
I'm seeing some very strange behavior where calling .status() on a wifi socket causes the socket to reset itself (turn off and then on immediately). It doesn't happen every time .status is called, but I can very very regularly see this happen. I've done some digging but haven't been able to diagnose why it's happening yet.

Any ideas as to what's causing this and how to fix it?

clach04, 2018-09-22:
This is sounds like #29 I've not seen this with my devices. It also sounds like https://github.com/codetheweb/tuyapi/issues/84 which is the Node implementation I'll post comments there.

glentakahashi, 2018-09-22:
Okay I'll close out in track in the other issues

closed 2018-09-22 by glentakahashi