clach04/python-tuya #38: Handing over maintenance of sean6541/tuya-homeassistant

sean6541, 2018-10-06:
I have made clach04 a contributor to my sean6541/tuya-homeassistant repository and an owner of the PyPi package pytuya. Feel free to reset the versioning of pytuya, but remember to also update the requirement version in sean6541/tuya-homeassistant/tuya.py

clach04, 2018-10-09:
Thanks @sean6541 for giving me permissions to:
 * PyPi - this helps with issue #34 
 *  https://github.com/sean6541/tuya-homeassistant in relation to #24

I was able to upload a new version 7.0.1 with the current code to PyPi.

I'm not sure if I'll get chance to hack on the HA support but I'm happy to assist with reviewing Pull Requests.

clach04, 2018-10-09:
NOTE I'm not sure resetting versions is possible in PyPi.

closed 2018-10-09 by clach04

jchulce, 2018-12-15:
Hey, where did tuya-homeassistant go? The repository now 404's.