clach04/python-tuya #39: pip install pytuya fails

slomanl1, 2018-10-09:
When I try to install pytuya package I get the following error:
pip install pytuya
Collecting pytuya
  Using cached https://files.pythonhosted.org/packages/81/7a/a77347f79ab92408acf49d48d15043fa1254405fb501f2ecc6e64f2c7ca9/pytuya-7.0.1.zip
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-yfyZYU/pytuya/setup.py", line 23, in <module>
        with codecs.open(os.path.join(here, 'README.md'), encoding='utf-8') as f:
      File "/usr/lib/python2.7/codecs.py", line 896, in open
        file = __builtin__.open(filename, mode, buffering)
    IOError: [Errno 2] No such file or directory: '/tmp/pip-install-yfyZYU/pytuya/README.md'

cheshirekow, 2018-10-15:
Yeah, same here. I tried python2 and python3. I've got `pip 18.1` if it matters. Looks like a simple packaging error though (`README.md` not in the package)?

Version `7.0` still installs fine:
```
pip install --user pytuya==7.0`
Collecting pytuya==7.0
  Downloading 
...
Installing collected packages: pyaes, pytuya
Successfully installed pyaes-1.6.1 pytuya-7.0
```

clach04, 2018-10-16:
Thanks for the heads up.

clach04, 2018-10-16:
Should be resolved by d9e72d335b5f234004ef962628a598bdd87613ed - will close in a week.

clach04, 2018-10-16:
https://pypi.org/project/pytuya/ version 7.0.2 posted to PyPi

closed 2018-10-20 by clach04