clach04/python-tuya #40: Missing CRC32

SynAckFin, 2018-10-21:
Thanks for this.
I've been looking at the communications with my device and the suffix isn't "000000000000aa55".
It is CRC32 + "0000aa55". The CRC32 is of everything before (prefix -> payload) and is stored as a big endian.
The demo code still works with the CRC set to zero but they may decide in future (or with some devices) to reject anything with an incorrect CRC.
My device is labelled TECKIN WiFi Smart Plug

clach04, 2018-11-13:
@SynAckFin that makes more sense than blank padding, good catch :-)

Have you tried sending CRC32 instead of blanks or is this based on seeing comms from other clients?

BillSobel, 2018-11-13:
My client correctly sets the CRC....

belzedaar, 2019-07-08:
This is set correctly in latest version