clach04/python-tuya #44: How to run demo?

melgard1, 2018-11-10:
New to python. Have installed python-tuya on Raspberry Pi using
`python -m pip install python-tuya`

I copied the demo code into demo.py and tried to run it.

```
python demo.py
Traceback (most recent call last):
  File "demo.py", line 7, in <module>
    d = pytuya.OutletDevice(deviceid, ipaddress, key)
AttributeError: 'module' object has no attribute 'OutletDevice'
```

dmelgar, 2018-11-10:
FYI: Was caused by running against python v2.7.13. I switched to using `python3`, version 3.5.3 and its working great!

dachshund-digital, 2019-06-27:
Same issue with my attempt.  I installed using # pip3 install python-tuya.  No errors.  I am running python 3.5.3.