clach04/python-tuya #49: BHT-6000 Thermostat (Decrypt)

fraschizzato, 2019-01-09:
I have two af that devices, i'm using the following code to get data from both:

```
import socket
import struct
import base64
import time

import pytuya

key = {'<local_key1>' , '<localKey2>'}

for k in key:
  print k
  
host_port = 6666
host_ip = '239.255.255.250'

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
sock.setsockopt(socket.IPPROTO_IP, socket.IP_MULTICAST_TTL, 2)
sock.bind(('', host_port))

mreq = struct.pack('4sl', socket.inet_aton(host_ip), socket.INADDR_ANY)
sock.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, mreq)

try:
    while True:
        print('listening')
        data = sock.recvfrom(1024)
        raw_bytes, peer_info = data
        print(data)
        packet = raw_bytes
        PREFIX_LEN = 16
        SUFFIX_LEN = 8
        prefix = packet[:PREFIX_LEN]
        json_crypted_plus_suffix = packet[PREFIX_LEN:]
        json_crypted_len = ord(prefix[-1])
        json_crypted_b64_with_header = json_crypted_plus_suffix[:json_crypted_len-SUFFIX_LEN]
        version_str = json_crypted_b64_with_header[:3]
        print('version_str (%d) %r' % (len(version_str), version_str))
        hexdigest = json_crypted_b64_with_header[3:][:16]
        print('hexdigest (%d) %r' % (len(hexdigest), hexdigest))
        json_crypted_b64 = json_crypted_b64_with_header[3+16:]
        suffix = packet[-SUFFIX_LEN:]  # assume we know length of end

        print('packet (%d) %r' % (len(packet), packet))
        print('prefix (%d) %r' % (len(prefix), prefix))
        print('json_crypted_plus_suffix (%d) %r' % (len(json_crypted_plus_suffix), json_crypted_plus_suffix))
        print('json_crypted_len %d' % (json_crypted_len))
        print('suffix (%d) %r' % (len(suffix), suffix))

        print('json_crypted_b64 (%d) %r' % (len(json_crypted_b64), json_crypted_b64))

        for k in key:
          try:
            print k
            cipher = pytuya.AESCipher(k)
            json_raw = cipher.decrypt(json_crypted_b64)
            print('json_raw (%d) %r' % (len(json_raw), json_raw))
          except:
            print 'Error'
            continue
        print '#################################################################'
finally:
    sock.close()
```
 But in execution I get error on decrypt:

```
listening
('\x00\x00U\xaa\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x9f\x00\x00\x00\x00{"ip":"192.168.1.250","gwId":"<dev_id>","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"IAYz2WK1th0cMLmL","version":"3.1"}\x12\xb8\xac\xb8\x00\x00\xaaU', ('192.168.1.250', 49154))
version_str (3) '\x00\x00\x00'
hexdigest (16) '\x00{"ip":"192.168.'
packet (175) '\x00\x00U\xaa\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x9f\x00\x00\x00\x00{"ip":"192.168.1.250","gwId":"<dev_id>","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"IAYz2WK1th0cMLmL","version":"3.1"}\x12\xb8\xac\xb8\x00\x00\xaaU'
prefix (16) '\x00\x00U\xaa\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x9f'
json_crypted_plus_suffix (159) '\x00\x00\x00\x00{"ip":"192.168.1.250","gwId":"<dev_id>","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"IAYz2WK1th0cMLmL","version":"3.1"}\x12\xb8\xac\xb8\x00\x00\xaaU'
json_crypted_len 159
suffix (8) '\x12\xb8\xac\xb8\x00\x00\xaaU'
json_crypted_b64 (132) '1.250","gwId":"<dev_id>","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"IAYz2WK1th0cMLmL","version":"3.1"}'
<local_key1>
Error
<local_key2>
Error
```

With tuyaAPI i can get actual data and set parameters:

```
{ devId: '<dev_id>',
  dps:
   { '1': true,
     '2': 32,
     '3': 36,
     '4': '1',
     '5': false,
     '6': false,
     '102': 27,
     '103': '0',
     '104': false } }
```

Notes on that device:
- In automatic mode the device does not report "temperature_set" correctly, it remains at previous manual set (Try to get all data to find out if data is not sended to tuya or not generated at all)
- The device does not report if triggered or not (heat on / off) (same as previous)
- Finding if that device can be updated with (https://github.com/xoseperez/espurna)
- Google home integration (only commands supported by default) can be full by homeassistant integration (https://community.home-assistant.io/t/bht-6000-thermostat-tuya-support/89311)

dps updated in wiki

BillSobel, 2019-01-09:
Can you post a link to the device or a model number?

belzedaar, 2019-07-08:
Maybe it's protocol version 3.3? I pull request just went in to support 3.3 based on the javascript code in the tuyaAPI repo