clach04/python-tuya #53: ImportError: No module named 'pytuya'

RR2301, 2019-06-14:
I am trying to run python program in raspberrypi3. but it gives following error.
`
pi@raspberrypi:/usr/lib/python3.5 $sudo python3 Test2.py
Traceback (most recent call last):
  File "Test2.py", line 3, in <module>
    import pytuya
ImportError: No module named 'pytuya'
`
Just FYI, I already installed pytuya module
`
pi@raspberrypi:/usr/lib/python3.5 $ pip3 install pytuya==7.0.2
Collecting pytuya==7.0.2
  Using cached https://www.piwheels.org/simple/pytuya/pytuya-7.0.2-py3-none-any.whl
Collecting pyaes (from pytuya==7.0.2)
  Using cached https://www.piwheels.org/simple/pyaes/pyaes-1.6.1-py3-none-any.whl
Installing collected packages: pyaes, pytuya
Successfully installed pyaes-1.6.1 pytuya-7.0.2
`
Appreciate your quick help!
Note : My overall objective is to control tuya switch from Openhab panel.

alex-phillips, 2019-06-15:
Also getting this error in my Home-Assistant setup

clach04, 2019-06-16:
@RR2301 from your report it doesn't look like the module is installed globally for the root user. It looks like it is only installed locally.

Try installing for root, e.g.

    sudo pip3 install .....

alex-phillips, 2019-06-16:
My issue was there was a change in the Home-Assistant software in how I specified required modules. So it was also a pip user install issue for me, now fixed.

Love the project, BTW. Great job!

qntris, 2020-03-23:
@alex-phillips , can you elaborate a bit more on this - I am also getting that import error for "No module named 'pytuya'", although the module is in the same folder. Can you share your steps for making it work?