clach04/python-tuya #54: New tuya 3.3 version

Jalleronline, 2019-06-18:
Hello, first of all thank you for your work.

I am having problems into getting the payload and decrypting it in the new 3.3 protocol version.

This problem was reported and discussed in this topic of tuyapi:
https://github.com/codetheweb/tuyapi/issues/175

Have you got plans for update your python library in order to be compatible with the new 3.3 version?

Regards.

belzedaar, 2019-07-08:
I got 3.3 working in my fork and it's now been integrated back. Give it a try and let me know if it doesn't work

sunwire, 2019-07-08:
I confirm, it works now.

Jalleronline, 2019-07-08:
Same here, works like a charm.
Thank you very much.

mobamoba, 2019-07-10:
I can't seem to get 3.3 to work with the most recent version. After entering data = d.status() as per the example, I get the following. Any idea how to fix? Thanks.
Unexpected status() payload=b'3.3\x00\x00\x00\x00\x00\nw\xbf\x00\x00\x00\x01\xd2\x97Q1X\xd5[\xeb\xb6
\xa5k\x04{\xe6\xfe\x1e\x10\xb4\x0e\x0c<\xbd*G\x05\x9c[;\xe0\x12\x98\xa4]\x02\xd4\x14/\x87\xe4g\x1c\x
1e\xfc\xee\x87|\x9e\x9d9\x94A\xa3\x0c\x9c\x03\x94\xf4\xc3\xfe\xc9\xed\x18N\xaa\xd6Ac\xf3\x9a\x13u\xc
5"\x98MW\xe8\xd4\xb0\x1f'

belzedaar, 2019-07-13:
Can you pastbin the code?

mobamoba, 2019-07-14:
Sorry, what code?

belzedaar, 2019-07-15:
Whatever code you're using that calls pytuya.  For version 3.3 you need to call d.set_version(3.3) before calling .status() for example.

mobamoba, 2019-07-15:
Ah - I was just using the sample code on the front page of this git, which I guess hasn't been updated to note that yet. Did that and it works fine thanks

cschutte, 2019-08-18:
I also have problems to get status on 3.3 devices. To try and isolate the problem that version 3.3 is actually used, I had hard coded the version into the status(self) function, by adding line "self.version = 3.3"at the top of the function. (I believe my devices are using 3.3. When I set the version to 3.2, it stops working completely.)

The code traceback as follows
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 249, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 419, in async_device_update
    await self.hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/local_tuya/switch.py", line 158, in update
    status = self._device.status()
  File "/config/custom_components/local_tuya/switch.py", line 116, in status
    self._cached_status = self.__get_status()
  File "/config/custom_components/local_tuya/switch.py", line 94, in __get_status
    status = self._device.status()
  File "/config/custom_components/local_tuya/pytuya/__init__.py", line 311, in status
    result = cipher.decrypt(result, False)
  File "/config/custom_components/local_tuya/pytuya/__init__.py", line 87, in decrypt
    raw = cipher.decrypt(enc)
  File "/usr/local/lib/python3.7/site-packages/Crypto/Cipher/_mode_ecb.py", line 195, in decrypt
    raise ValueError("Data must be aligned to block boundary in ECB mode")
ValueError: Data must be aligned to block boundary in ECB mode