clach04/python-tuya #55: Cannot use module, OutletDevice method does not exist

dachshund-digital, 2019-06-27:
I installed using # pip3 install python-tuya. No errors. I am running python 3.5.3. Tried to list classes via import, dir method, no classes shown.

pip3 install python-tuya

python
Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.

import pytuya
dir(pytuya)
['builtins', 'cached', 'doc', 'file', 'loader', 'name', 'package', 'path', 'spec']

NLFrankG, 2019-07-13:
Same issue here after updating to 7.0.2 version of pytuya to get protocol 3.3 support using python 2.7.9. All worked well before updating using #  pip install python-tuya

rbpaul, 2019-12-17:
How does one revert to 7.0?

chsims1, 2020-05-04:
`AttributeError: module 'pytuya' has no attribute 'OutletDevice'`

I understand that this is not actively maintained, but I am surprised that there have been no comments on this issue.  The latest version works fine on python 2.7.16 but not on 3.7.3.  Obviously I'd like to not have to stick with deprecated python 2.