clach04/python-tuya #62: 7.0.4 release on PyPI

Hedgehog57, 2019-09-26:
Hello,

Thanks for a good work on supporting local access to Tuya devices. 

On https://pypi.org/project/pytuya/ only version 7.0.2 is released. Currently many Tuya devices require 3.3 version of protocol.. It is supported only in verison 7.0.4 of pytuya.

In Home Assistant it is very hard to use import not from packages released from pypi.org.

Please do something with it. Release version 7.0.4 on pypi.org.

Thanks.

clach04, 2019-09-28:
@Hedgehog57 I just pushed the release to https://pypi.org/project/pytuya/7.0.4/. As a heads up; The sanity checks in the tests are failing, it appears to be due to suffix sanity checks - I'm assuming the sanity check is too strict. I'm back to using x10 devices (been more reliable for me than these devices) so I'm not able to test. with real hardware.

closed 2019-09-28 by clach04

Hedgehog57, 2019-09-28:
Thank You! Now it's much easier to use pytuua in custom plugin for Home
Assistant. Especially in HASSIO. There is no good way to install not
published libraries.