clach04/python-tuya #64: new switches different port

Jefronty, 2019-11-14:
My older plugs (KMC) broadcast to port 6666 both from 49155.  My new Tuya/Smart Life plugs (Teckin) both broadcast to port 6667 from 49154.  
I can get the status and control the older plugs using pytuya, but whenever trying to interact with the newer plugs with python, even retrieving the status I get an error.  I'd guess that maybe these new plugs are listening on a port other than 6668, but I don't know which one.  Does anyone know how to communicate with these plugs?  
The error output when trying to get the status is:  
```  File "<stdin>", line 1, in <module>                                                                  │Generating grub configuration file ...
  File "/usr/local/lib/python2.7/dist-packages/pytuya/__init__.py", line 281, in status                │Found linux image: /boot/vmlinuz-4.15.0-70-generic
    data = self._send_receive(payload)                                                                 │Found initrd image: /boot/initrd.img-4.15.0-70-generic
  File "/usr/local/lib/python2.7/dist-packages/pytuya/__init__.py", line 179, in _send_receive         │Found linux image: /boot/vmlinuz-4.15.0-66-generic
    data = s.recv(1024)
```

clach04, 2019-11-15:
@Jefronty thanks for sharing, that is intereting. I would check in with https://github.com/codetheweb/tuyapi they have been actively updating protocol support. I'm not following it closely but I thought I saw a discussion about a new protocol seen with some devices.

Jefronty, 2019-11-15:
@clach04 Thanks for the response.  I did check [TuyApi](https://github.com/codetheweb/tuyapi) and wrote a quick test js script to try.  Using the parameter version=3.3 when adding the device worked.  I think the best resolution would be adding an optional version parameter to the OutletDevice class in pytuya, from there I don't know what other changes would need to be made.  If the new outlets can't be controlled with python, at least I can use subprocess or something to check or toggle them.  I may look into the differences that the version does in TuyApi and if I figure out how to mimic them in python, I'll submit a pull request.

doubledrat, 2019-12-26:
I'm having the same problem with teckin smart plugs bought in December 2019