clach04/python-tuya #67: PIR & Door/Window Sensor Support?

codeclinic, 2020-01-08:
Does this support the usage of PIR &/or Door/Window Sensors?

clach04, 2020-01-18:
No - its mostly for polling WiFi devices - I suspect those sensors actively send data rather than being polled (somehow probably via the hubs I've seen online (I got the impression there are some 433Mhz sensors that connect to a hub rather than wifi).

closed 2020-01-18 by clach04

BillSobel, 2020-01-18:
Sensors send data via Mqtt only.

codeclinic, 2020-01-18:
> Sensors send data via Mqtt only.

Sensors send data via MQTT? Really? Do you have any info on how we can catch this data?

BillSobel, 2020-01-21:
Its all documented in the Tuya SDK, but you need actual API keys to login
to MQTT and access the data you cant grab that from the devices..