clach04/python-tuya #68: d.status timeout with a just bought device

puppinoo, 2020-01-22:
Hi,
I have an older different device working great.
I just bought this one https://www.amazon.it/gp/product/B07X2S13N8 cause they had italian plugs. Been able to retrieve keys using older smartlife and tuya-cli for both of them. Unfortunately usual test code fails at d.status line. The same code works fine with older device. Any advice?
Thanks.
Pino.

 `python3.7  tuyacmd2.py 
Traceback (most recent call last):
  File "tuyacmd2.py", line 6, in <module>
    data = d.status()  # NOTE this does NOT require a valid key
  File "/www/cgi-bin/protected/tuya/python-tuya/new/pytuya/__init__.py", line 279, in status
    data = self._send_receive(payload)
  File "/www/cgi-bin/protected/tuya/python-tuya/new/pytuya/__init__.py", line 177, in _send_receive
    data = s.recv(1024)
socket.timeout: timed out
`

puppinoo, 2020-01-31:
Update: setting version to 3.3 made d.status work but not when I try to set DSP 1 I receive:
`Dictionary {'devId': 'xxxxxxxxxxxxxx', 'dps': {'1': True, '9': 0, '18': 0, '19': 0, '20': 2333, '21': 1, '22': 595, '23': 27972, '24': 15484, '25': 1345}}
state (bool, true is ON) True
Traceback (most recent call last):
  File "tuyacmd2.py", line 12, in <module>
    data = d.set_status(not switch_state)  # This requires a valid key
  File "/www/cgi-bin/protected/tuya/python-tuya/new/pytuya/__init__.py", line 329, in set_status
    data = self._send_receive(payload)
  File "/www/cgi-bin/protected/tuya/python-tuya/new/pytuya/__init__.py", line 177, in _send_receive
    data = s.recv(1024)
ConnectionResetError: [Errno 104] Connection reset by peer`

dharichandan, 2020-03-01:
Did you resolve it ???

puppinoo, 2020-03-01:
Yes, I implemented a retry logic found on the forum: Maybe this one
https://github.com/clach04/python-tuya/pull/43/files/cb236da4aa38fa398f238adc1e12003e72114094..f624ef03490995c6780ffa7edccfccc9d885a3a8#diff-85460c06e57fa5b9a18b00b76b19c5bf

dharichandan, 2020-03-04:
Hi I used the same but still not able to run device. One question - 
How did you extract the deviceKey ? 
I have deviceId, IP and port = 6888. 
But device Key is something hard to find and my firmware is 1.0.1 does that affect way I use the library?

puppinoo, 2020-03-04:
> Hi I used the same but still not able to run device. One question -
> How did you extract the deviceKey ?
> I have deviceId, IP and port = 6888.
> But device Key is something hard to find and my firmware is 1.0.1 does that affect way I use the library?

I used the usual procedure (proxy configured on my celphone + tuya-cli on my linux pc) BUT I had to:
1) Install older Smart Life app on my android phone (3.12.6 version. Find on the web) and sisable updates on google play for this app
2) Disable ipv6 on my ubuntu pc 
3) Disable data on my cellular.

Without these steps otherwise nothing worked.

Regards.

dharichandan, 2020-03-04:
Thanks. I will try it out. But here are you capturing packets ? Or do you have any link for step by step procedure to setup tuya-cli and also extract keys?

puppinoo, 2020-03-04:
> 
> 
> Thanks. I will try it out. But here are you capturing packets ? Or do you have any link for step by step procedure to setup tuya-cli and also extract keys?
> 
> This is something I found -
> https://github.com/TuyaAPI/cloud

https://www.apkmirror.com/apk/tuya-inc/smart-life-smart-living/smart-life-smart-living-3-12-6-release/smart-life-smart-living-3-12-6-android-apk-download/

https://github.com/codetheweb/tuyapi/blob/master/docs/SETUP.md

puppinoo, 2020-03-04:
you need the "Linking a Tuya Device with MITM (deprecated)"