clach04/python-tuya #71: 7.0.6 release

nikrolls, 2020-05-12:
Could we please get a new release? The work since `7.0.5` fixes pip installs which I need for my Home Assistant integration, and even updates the version number to `7.0.6`, but there hasn't been a release to pypi meaning I can't use it.

clach04, 2020-05-12:
@nikrolls I posted https://github.com/clach04/python-tuya/issues/71 this is 100% **un**tested.

I'm pretty sure HA extension.REQUIREMENTS list is piped through `pip` so assuming the machine has git installed one can use regular git URLs (see https://pip.pypa.io/en/stable/reference/pip_install/#git )which avoids the urgency to publish (especially whilst developing). Something I've NOT tried with pip is using download links like https://github.com/clach04/python-tuya/archive/master.zip

closed 2020-05-12 by clach04

nikrolls, 2020-05-12:
This may work, but currently I'm attempting to have my integration's dependencies preloaded in HA, and based on the HA requirements it needs to come from PyPI. Unless that's different for the wheel repository, @frenck?